malariaspot/MicroSpot-WiFi #13: When several jogAxis petitions are made, ESP8266 overflows and restarts.

gvJaime, 2017-05-16:
It has been confirmed that the whole system restarts, by putting a homeAxis() command in the Mechanical Toggle routine.

This is a very important issue because jogging is used for panning: one of the most common user operations.

alexx666, 2017-05-24:
As we already saw its a memory problem, which is currently partly solved. 
Pending: String parsing and stress tests

gvJaime, 2017-05-26:
torture.py tests reveal that, if the request spamming has a period < 1second, the ESP8266 overflows after ~195 requests.

It doens't restart with the python script, but it restarts with a phone.

The python script failure is limited to just the esp8266 returning empty responses, although commands can be sent. After a while it sends the right responses again.

alexx666, 2017-05-26:
Another simple test we could run is try to overflow the esp8266 by fluding it with /ayy/lmao requests. Test it with deferent speeds. Each second, each 500ms, 250ms, 100ms etc...

That would help us narrow down the memory bottle neck. If it fails in any way, it would mean that either we dont manage memory correctly in MicroServer or WiFiServer simply has limitations to how many requests it can handle.

On the other hand if it doesnt restart or fail that means Mechanical.ccp has the memory problem, because all other requests, appart from /ayy/lmao, require it to be called.

gvJaime, 2017-05-29:
Adding a ```delete client``` after ```client->stop()``` breaks the OTAs

gvJaime, 2017-11-20:
Won't fix

closed 2017-11-20 by gvJaime