malariaspot/MicroSpot-WiFi #25: use Serial1 driver for Uno WiFi. access the esp at 230400 baud.

jandrassy, 2017-12-13:
Implements[ Serial1 of Arduino Uno WiFi Developer Edition board](https://github.com/jandrassy/UnoWiFiDevEdSerial1) to access the on-board ESP8266 at 230400 baud.

The library is a driver for SC16IS750 implemented for Arduino Uno WiFi Developer Edition as Serial1. This driver is necessary to communicate with the on-board ESP8266 over it's serial interface. The included tool EspProxy enables accessing the on-board ESP8266 at 115200 baud over USB for 'flashing' tools (esptool, FDM), IDE sketch upload and Serial Monitor.

jandrassy, 2017-12-13:
I created the driver, but at the beginning some months ago, it was your Uno WiFi know-how, what helped me to start explore the Uno WiFi possibilities

gvJaime, 2017-12-14:
[image] [Responder al issue de Microspot-WiFi](https://trello.com/c/8GOlcEa5/1-responder-al-issue-de-microspot-wifi)

gvJaime, 2018-02-13:
Hello @jandrassy 

We're glad that our repo helped you make the best from your ESP8266 board. It is always a pleasure to contribute to a new development.

Just by curiosity, how did our repo help you in your project?

Also, is this an issue within our firmware? if not, should I close this issue?

jandrassy, 2018-02-13:
Sorry, no issue with your firmware, only a way to communicate. Your [Wiki page about Arduino Uno WiF](https://github.com/malariaspot/MicroSpot-WiFi/wiki/Arduino-UNO-WiFi-ESP8266-Arduino-core-compatibility-notes)i was the help.
And the library I created for the Uno WiFi could help you for fast communication between Atmega and esp8266 on Uno WiFi.

closed 2019-03-01 by jandrassy